jotego/jtbubl #2: Not enough Bubbles in Bubble Bobble intro

Chocobobobo, 2020-08-14:
First of all, this bug is courtesy of Dampfwalze.
During the intro, you have bubbles moving around in circle motion in the original, but only 2 bubbles going up vertically in the current version of the core.
Proper behavior:
https://youtu.be/1dadJO4YtMk
[image]

Current MiSTer behavior
https://youtu.be/n6UfKNzD0kI
[image]

Jotego mentioned it might be MCU related, but we will let the pro take it from there! :)

Brettster247, 2020-08-15:
I have noticed that in the US set 5.1 you can select normal or super mode. In normal mode just the 2 bubbles, but in Super Mode you get all the bubbles during the intro.

jotego, 2020-12-17:
I have improved the accuracy of the timing of all CPUs in commit https://github.com/jotego/jtbubl/commit/03ae93917475f82f8e4eecfed893bf2f9f3b4ae2. But it made no difference to this issue. 
6801 speed could still be wrong as I do not know whether that core is cycle accurate.

PhantombrainM, 2021-12-20:
It seems to be fixed? I see all the bubbles now. 
Only red intro text is a little flickery due to the bubbles?

jotego, 2021-12-20:
You're right! Maybe I fixed inadvertedly while working on the Tokio flip mode. I see the bubles in version 5fd9001. I'll check the red text problem. Thanks a lot!

atrac17, 2021-12-20:
Accidental Accuracy! It's a Christmas miracle!!! Nice find @PhantombrainM.

Chocobobobo, 2021-12-20:
A Xmas miracle indeed... I can't check the fix but I will trust my twin for sure. Thanks Phantom. If only Black Dragon could get the same miracles in the late stages...

jotego, 2021-12-20:
As far as I know, Black Tiger doesn't hang up anymore.

Chocobobobo, 2021-12-21:
There are other problems, GFX glitches. But I am sure MJ is already on it, chasing the real PCB in japanese arcades to confirm those, like me.

jotego, 2021-12-27:
Fixed in 23a356f
I'll update the binaries in a couple of hours

closed 2021-12-27 by jotego

gizaha, 2022-01-16:
In non super mode, it checks "rng" (R register, beam's H-pos) in order to calculate the first bubbles, 
Without it, it doesn't work. Rom bublbobl.zip,  breakpoint 268C

jotego, 2022-01-16:
Thank you, it is a very informative comment. I understand you refer to the Z80's R register -refresh. I believe it is correctly implemented in the T80 core, although I have not been able to find the specific lines that take care of it. CPU registers in VHDL tend to be a bit hidden under strange names and bit encodings, so it is generally not straight forward to find them.
I think the bubbles are correctly displayed now and the problem was related to the GFX system in my case.