jotego/jtbubl #5: Support for DB9 on MiST-Board

uchristo, 2020-08-26:
I‘m currently unable to use the DB9-Ports of my MiST for the BubbleBobble Core. Is this intended? I connect a 2-Button-Version of Competition Pro-Series.

A Gamepd connected through USB works (in the test with the DB9, there was no USB Pad connected)

jotego, 2020-08-26:
I am actually not familiar with configuring DB9 devices in MiST but I know that the joystick submodulw for MiST follows the original one provided by Gyurco. I'll invite him in case he can suggest a solution.

gyurco, 2020-08-26:
It's handled by the firmware, you have nothing to do to make it work. However they're shifted one position if an USB gamepad is detected. Without any USB stick plugged in, joystick0/1 signals are the two DB9 ports. If one USB gamepad present, then one DB9 is usable. With two gamepad, only joystick3(4,5) could be the DB9 port.

jotego, 2020-08-27:
Thank you so much for that suggestion. I actually started implementing that with the JOIN_JOY macro for games that share the same joystick for first and second player but thought it wasn't very useful. 
I'll make a point in adding it to single player games.

For Bubble Bobble the solution is simply to disconnect the USB joystick or play with the second player then.

uchristo, 2020-08-27:
Okay, I just tested and had to set the option joystick_db9_fixed_index=1 in mist.ini in order to get it working.
There's just one caveeat at the moment:
I need to press Button2 for jumping but would like to use "Direction Up" instead (like on Amiga conversion of this game). Unfortunately, Joystick-Remapping seems to be available just for USB-Devices.

jotego, 2020-08-27:
Maybe @gyurco can give you a generic solution for MiST. 

As for this particular core, I think it's reasonable to add an OSD option for that if it cannot be done in a generic way within the MiST firmware.

gyurco, 2020-08-27:
There's no option in the firmware to remap the DB9 directions. Feel free to send a patch if you need it :)
The code which handles DB9:
https://github.com/mist-devel/mist-firmware/blob/master/user_io.c#L911

uchristo, 2020-08-28:
I'd now try to reuse the USB_Remap-Feature and create a string with VID=0X00db and PID 0x0000 and 0x0001 to indicate I want to remap the DB-9 Ports. Mapping would be the DB9-Pinout (PIN1: UP, PIN2: DOWN, PIN3: LEFT, PIN4: RIGHT, PIN9: FIRE1, PIN6, FIRE2
Regarding default_joystick_mapping [16] for the first Port the string would look like: 0x00db,0x0000,4,3,1,2,9,6,0,0,0,0,0,0,0,0,0,0

gyurco, 2020-08-28:
I would not protest against a db9_remap option.

retrofun, 2020-08-28:
No need to reinvent the wheel. With the existing joystick mapping framework it's a 3 line change:

https://github.com/retrofun/mist-firmware/commit/d95bb2ea8ad87efcbaaa40a43d9cae86e8f7a034

Then you can remap like explained in the wiki: https://github.com/mist-devel/mist-board/wiki/USBJoystickMapping

BTW: just noticed that since firmware 200614 the DB9 joystick tests in OSD are broken when USB joysticks are connected. Probably caused by ```- Add joystick_db9_fixed_index (by PunisherHD92)```

jotego, 2020-08-28:
Official DB9 support is lacking in MiSTer so improving it in MiST is a competitive point.

I can try giving this some visibility on Twitter. Let's wait till Monday as people browse twitter more on working days.

uchristo, 2020-08-28:
> No need to reinvent the wheel. With the existing joystick mapping framework it's a 3 line change:
> 
> [retrofun/mist-firmware@d95bb2e](https://github.com/retrofun/mist-firmware/commit/d95bb2ea8ad87efcbaaa40a43d9cae86e8f7a034)
> 
> Then you can remap like explained in the wiki: https://github.com/mist-devel/mist-board/wiki/USBJoystickMapping
> 
> ```
> [bublbblr]
> joystick_remap=00db,0001,1,2,4,20,10
> ```
> 
> BTW: just noticed that since firmware 200614 the DB9 joystick tests in OSD are broken when USB joysticks are connected. Probably caused by `- Add joystick_db9_fixed_index (by PunisherHD92)`

I‘m stunned..... tried to understand how everything is connected within firmware this morning. Would have continued on monday. The solution is really straight forward. I‘ll compile on monday and will give it a try. Stay tuned!

gyurco, 2020-08-28:
> BTW: just noticed that since firmware 200614 the DB9 joystick tests in OSD are broken when USB joysticks are connected. Probably caused by `- Add joystick_db9_fixed_index (by PunisherHD92)`

I think it's because of the joystick renumber rules when only 1 USB stick is present (check the joystick_renumber() function). My opinion is those renumbering rules became so confusing that they should be all removed.
I've already opened an issue for this:
https://github.com/mist-devel/mist-firmware/issues/37

retrofun, 2020-08-29:
Found the missing bit... Hopefully I got it right:

https://github.com/retrofun/mist-firmware/commit/74325b2b1072445885df0024d397338a83c274e0

And I agree, the joystick code with reordering is confusing.

gyurco, 2020-08-29:
The patches look good and simple, I think you should send a PR.

retrofun, 2020-08-30:
Patches are merged in MiST firmware repository, https://github.com/mist-devel/mist-firmware/commits/master.

jotego, 2020-08-30:
Great news. I guess we just need confirmation from @uchristo to close the issue

uchristo, 2020-08-31:
I can confirm that the remapping is working! Thanks a lot for the very fast implementation!

closed 2020-08-31 by jotego