jotego/jtbubl #6: Has stopped working?

brimoran, 2020-09-28:
I just get a black screen for Bubble Bobble and Tokio Scramble.  Possibly due to updating MiSTer as Bubble Bobble at least was previously working fine.  However I used the update_all script so a few things updated at the same time so I can’t be sure.

I’m on MiSTer version 200922 and the rbf is 20200815.

I can see a similar report on the Jotego discord.

jotego, 2020-09-28:
Hi Brimoran,

It definetely hasn't stopped working. Long story short, MiSTer cannot find your zip files for some obscure reason. I was facing a similar problem yesterday myself while working on new MRA files. 

I suggest you look around for differences between your setup -particularly about mame folders- and what MiSTer firmware developers expect. They may have changed something about the mame zip search process in this release, or one of the script messed with the folders.

I'm sorry the only thing I can do is updating and testing the game. I did, and it still works with latest MiSTer files.

closed 2020-09-28 by jotego

brimoran, 2020-09-28:
OK thanks Jotego I’ll have another look.  Sorry for wasting your time!