jotego/jtbubl #15: Bubble bobble core Turbo option possible?

suww37, 2022-06-09:
Hello~ Thanks to you, My game life is always fun. In the bubble bobble core, the number of bubbles and the number of enemies increases, the delay is. This is not the case in the MAME emulator. In the past, you could have more fun by adding the Turbo function to Double dragon 1 and 2, is it possible to add the Turbo function to Bubble bobble? I hate delays in games. Please support Turbo as an option for Bubble bobble core. Thank you.

jotego, 2022-06-13:
For most people, the slowdown is preferred because of accuracy but I understand your point too. It's not always straight forward to speed up the games. But I will check whether it is possible without messing the logic too much.

jotego, 2022-12-02:
CPU clock recovery was disabled in this core. That's probably what causes the slow down.

- [ ] Check that the core is stable in MiST
- [ ] Check that the core is stable in MiSTer
- [ ] Update JTBIN