jotego/jtbubl #21: Schematic notes

jotego, 2022-11-28:
corrections for **exterm** schematics:
seen in git commit: 1d4d084

- [ ] SND_SELECT should be ~{SND_SELECT} on page 7/8
- [ ] pins I6 and I7 of C3 (PAL16L8) probably connect to other devices (not only to the pull-up resistor)
- [ ] pin A14 of C9 (27C512) cannot be floating. 
- [ ] Check if pins A14 and A15 if C9 and C11 are the same
- [ ] Rename MAIN_AD to BANK1
- [ ] Rename ~{RESET} to ~{SND_RST}
- [ ] Rename CFG4 to ~{SND_RST'} and disconnect from CFG bus
- [ ] There is no CFG5, reduce CFG bus to 3 bits
- [ ] Rename CFG bus to BANK_CFG
- [ ] Move down C3 so all wires enter straight (_el cable a I1 entra haciendo un quiebro_)

[image]

skutis, 2022-11-29:
It might be that they did not need the inputs to the PAL and reduced the equations from 
o13 = i8 & i9 & /i11 & /o15 + i3 & i8 & i9 & /i11 = i9 & /i11 & /o15 + i3 & i9 & /i11 
o14 = i5 & i7 & o15 + i6 & i7 & o15 = i5 & o15 + i6 & o15

I will check the PCB later.

closed 2022-11-30 by jotego

etriana85, 2022-11-30:
updated extermination schematic.
[exterm.pdf](https://github.com/jotego/jtbubl/files/10123794/exterm.pdf)