jotego/jtbubl #22: [insectx] schematic signals

jotego, 2022-11-30:
- [ ] Rename SBSWD[1:0] to  SBANK[1:0], these two bits configure the active ROM bank (page)
- [ ] Rename SBSWD2 to SET_LOCK1
- [ ] Rename SBSWD3 to SET_LOCK2
- [ ] Rename SBSWD4 to CNT_UP1
- [ ] Rename SBSWD4 to CNT_UP2
- [ ] Move the DIP switches to the same page as the YM2203
- [ ] Delete the wire mess and leave the connection by labels between U51 and U43
[image]

skutis, 2022-11-30:
I can fix it if you assign me.

jotego, 2022-11-30:
Thank you. Please pull the repository first as we have just fixed the library file (it should be a link to the file in JTKICAD but it was just a plane file somehow)

jotego, 2022-12-01:
The color signal indexes are messed up. The large resistors should connect to the small indexes. For instance, the 6kOhm resistor should connect to R0, because a change in R0 has little effect in the output due to the large impedance. 

The 1kOhm resistors are out of the binary sequence. They probably set a general brightness or tone for the image. I haven't seen references to that in the MAME docs so far. I suggest we do not count them in the color bus but give them a different name, like Rb, Bb, Gb where the b stands for brightness. That would be pins 1, 7 and 14.

[image]

skutis, 2022-12-01:
I will have a look at it tonight. When I think of it, the 1k did not make sense to me when I extracted the scehmatics from the PCB.

jotego, 2022-12-01:
This is how @etriana85 did it for Insector X. I'm not sure what the DC in R/G/B DC stands for, it may just be a left over from a tentative name.

[image]

jotego, 2022-12-06:
I have checked all changes. Thank you

closed 2022-12-06 by jotego