jotego/jtbubl #46: Extermination schematics: change signal names for PAL output

jotego, 2023-01-21:
Skutis brought our attention to a couple of signals called MAIN_Axx coming out of a PAL. That is misleading as they are not related to the main CPU address used in the name. One of them is the watchdog clear and the other is related to the video memory.

@etriana85 update the schematics with good signal names, please.

etriana85, 2023-01-23:
Issue #46 Resolved. The names of the indicated signals have been corrected, as well as others.
I attach images, they are marked.

etriana85, 2023-01-23:
[image]
[image]
[image]
[image]
[image]
[image]

etriana85, 2023-01-23:
I have updated the datasheet of the symbol 74ls367, it had a link not found message.
I have moved the snd clock to the snd sheet, and some other gates, close to where the signals are generated, for those gates.

closed 2023-01-23 by etriana85