jotego/jtbubl #47: HB generation and length

jotego, 2023-02-02:
Pins 22 (HBLK) and 23 (SYNC) are connected together on Extermination and Arkanoid 2. But the connection is different on Extermination X. It looks like HB and SYNC can be wired together to produce some logic (wired _and_?) but we don't know exactly the signals that each pin would produce individually.

We want to know these signals

- [ ] HB/SYNC output when wired together @etriana85 
- [ ] HB when isolated @etriana85 
- [ ] SYNC when isolated @etriana85 
- [ ] Document the results in the design files @jotego 
- [ ] Document the results in the schematics @skutis and @etriana85

etriana85, 2023-02-09:
I'm working on this.

etriana85, 2023-02-09:
-VSYNC/VB sale at ~30Hz.
-The output signals of the SETA seem to go at half frequency.
-The oscillator output moves between 1 and ~3V, it is malformed.
According to conversation with Jotego, the oscillator will be replaced today.
We do not have an ideal plinth, but more information on the dimensions was already requested at TECNITRON, and thus have certainty to buy it.
Yesterday I made some measurements of HB and SYNC, with the different instructions, but they always returned 0 (zero).

skutis, 2023-02-09:
It's strange that it is 30Hz. It should be ~60Hz at one of the pins and 15.6kHz at the other. Do you have a picture of the measurements?

skutis, 2023-02-09:
ok. Is there something wrong with the oscillator?

etriana85, 2023-02-09:
We have discussed that when you return from your trip, we will decide what to do.